Mattzobricks/MattzoControllers #63: BT trains with different motor types

Mattzobricks, 2023-09-28:
On behalf of user Max I create the following feature request:

Hi there,
I’m currently building a loco that uses two M-motors in the loco and a train-motor in the tender.
Problem is that the train motor is driving faster then the M-motors, so the tender is pushing the loco the whole time. I’m planing to use the sbrick as the reciever is there a way to define the MTC4BT file that the different ports are run with different speeds so I could adjust it, to have both of them run at the same speed?

Thanks, Max

Mattzobricks, 2023-09-28:
As many useful features, this is actually implemented in the MTC4PF. For each motorshield, a "configMotor" parameter can be set:

    msConf[0] = (MattzoMotorShieldConfiguration){
        ...
        .configMotorA = -1,
        .configMotorB = 1,
        ...

While the values for these parameters are usually 1 (forward), -1 (reverse) or 0 (unused), the value can actually be any floating point value. This configuration for example would make motor B turn in the opposite direction with half the power of motor A:

    msConf[0] = (MattzoMotorShieldConfiguration){
        ...
        .configMotorA = 1,
        .configMotorB = -0.5,
        ...

We should also add such a feature to the MTC4BT.

Hilbert70, 2024-10-05:
Does adding 'pwrPerc' to the BT configuration solve this problem? This is already implemented.

Hilbert70, 2024-10-24:
Probably needs to be fixed in:
```
void MCChannelController::SetTargetPwrPerc(int16_t targetPwrPerc)
{
    _targetPwrPerc = normalizePwrPerc(_config->IsInverted()
                                          ? targetPwrPerc * -1
                                          : targetPwrPerc);
}
```
And add an item in _config and add an config item in the json file.

closed 2024-11-15 by Mattzobricks