Mattzobricks/MattzoControllers #64: Credentials for MQTT

Mattzobricks, 2023-12-07:
Some users have demanded optional credentials for MQTT.
Example: https://mattzobricks.com/forums/topic/mqtt-credentials-in-network_config#post-6559

Hilbert70, 2023-12-07:
Just change the occurrence of:

`boolean connect(const char* id, const char* willTopic, uint8_t willQos, boolean willRetain, const char* willMessage);`

To:

`boolean connect(const char* id, const char* user, const char* pass, const char* willTopic, uint8_t willQos, boolean willRetain, const char* willMessage);`
   
I think the BT part already has a password or username somewhere and otherwise it is easley extendable.

Mattzobricks, 2024-05-21:
Thought about this feature request again... I believe the number of users who would like to have this feature is small. Very small. It might be easy to implement, but it complicates the setup for beginners even more and potentially leads to errors.

Closing issue.

closed 2024-05-21 by Mattzobricks