Mattzobricks/MattzoControllers #66: All units of a loco connected: unreliable check

Mattzobricks, 2024-01-28:
When multiple BLE units that belong to the same loco connect to the MTC4BT, there is check if all of these units are connected. If not, the lights blink, and the loco won't move.

The check is unreliable. It happens frequently that upon the connection of all BLE units, the light of one of the units still blinks.

Tested with three different trains which physical locos all build a logical logo. The trains all have two units (ICE (Sbrick), EST and FUX (both PU)).  The error pops up on each of the trains.

redrooster75, 2024-12-04:
Hi there, I came upon the same issue with two locomotives, each with two PU-Hubs:
On the first locomotive, one hub continues to blink while the other does not blink, The second hub can be controlled via Rocrail. 
On the second locomotive, both hubs continue to blink and cannot be controlled by Rocrail.
Is this issue solved with version 1.1?

Hilbert70, 2024-12-04:
@redrooster75 are your configuration files, valid json files? Please check and look at the serial console if all hubs are discovered?
Are you seeing the ‘lc’ commands on the serial console of all locomotives?
Sorry for the questions!

Hilbert70, 2024-12-04:
@redrooster75 thanks for your response, I think you are indeed dealing with this issue. As I do not have pu lights I will try to reproduce with the build in led of the pu hubs.
And this is issue is not solved yet. If I can reproduce it, I can try to solve it.

Mattzobricks, 2024-12-15:
I am now able to reproduce the error. I tested it three times in a row, and the result was always the same.

This video describes the procedure of how to reproduce it:
(will be added here)

This is the controller configuration that I used:

controller_config.json:
```
{
	"name": "MTC4BT-BUWIZZ-TEST",
    "pwrIncStep": 25,
    "pwrDecStep": 25,
	"espPins": [
	],
	"fn": [
	],
	"locos": [
	],
	"locoConfigs": [
		"/loco_4hubs.json"
	]
}
```

loco_4hubs.json:
```
{
    "address": 6337,
    "name": "FUX",
    "pwrIncStep": 100,
    "pwrDecStep": 100,
    "bleHubs": [
        {
            "type": "PU",
            "address": "90:84:2b:01:20:f8",
            "channels": [
                {
                    "channel": "A",
                    "attachedDevice": "motor",
                    "direction": "forward"
                },
                {
                    "channel": "B",
                    "attachedDevice": "light",
                    "pwrIncStep": 10,
                    "pwrDecStep": 20
                }
            ]
        },
        {
            "type": "PU",
            "address": "90:84:2b:21:71:46",
            "channels": [
                {
                    "channel": "A",
                    "attachedDevice": "motor",
                    "direction": "reverse"
                },
                {
                    "channel": "B",
                    "attachedDevice": "light",
                    "pwrIncStep": 10,
                    "pwrDecStep": 20
                }
            ]
        },
		{
			"type": "SBrick",
			"address": "84:2e:14:4f:d3:14",
			"channels": [
				{
					"channel": "B",
					"attachedDevice": "motor",
					"direction": "reverse"
				},
				{
					"channel": "D",
					"attachedDevice": "motor",
					"direction": "forward"
				}
			]
		},
        {
            "type": "BuWizz2",
            "address": "50:fa:ab:42:15:25",
            "channels": [
                {
                    "channel": "1",
                    "attachedDevice": "motor",
                    "direction": "reverse"
                },
                {
                    "channel": "2",
                    "attachedDevice": "motor",
                    "direction": "reverse"
                },
                {
                    "channel": "3",
                    "attachedDevice": "light",
                    "pwrIncStep": 50,
                    "pwrDecStep": 50
                },
                {
                    "channel": "4",
                    "attachedDevice": "light",
                    "pwrIncStep": 50,
                    "pwrDecStep": 50
                }
            ]
        }
    ],
    "events": [
        {
            "triggers": [
                {
                    "source": "rr",
                    "eventType": "fnchanged",
                    "identifier": "f1",
                    "value": "on"
                },
                {
                    "source": "loco",
                    "eventType": "dirchanged",
                    "value": "forward"
                }
            ],
            "actions": [
                {
                    "address": "90:84:2b:01:20:f8",
                    "channel": "B",
                    "pwrPerc": 50
                },
                {
                    "address": "90:84:2b:21:71:46",
                    "channel": "B",
                    "pwrPerc": 0
                }
            ]
        },
        {
            "triggers": [
                {
                    "source": "rr",
                    "eventType": "fnchanged",
                    "identifier": "f2",
                    "value": "on"
                },
                {
                    "source": "loco",
                    "eventType": "dirchanged",
                    "value": "backward"
                }
            ],
            "actions": [
                {
                    "address": "90:84:2b:01:20:f8",
                    "channel": "B",
                    "pwrPerc": 0
                },
                {
                    "address": "90:84:2b:21:71:46",
                    "channel": "B",
                    "pwrPerc": 50
                }
            ]
        },
        {
            "triggers": [
                {
                    "source": "rr",
                    "eventType": "fnchanged",
                    "identifier": "f3",
                    "value": "off"
                }
            ],
            "actions": [
                {
                    "address": "90:84:2b:01:20:f8",
                    "channel": "B",
                    "pwrPerc": 0
                },
                {
                    "address": "90:84:2b:21:71:46",
                    "channel": "B",
                    "pwrPerc": 0
                }
            ]
        }
    ]
}
```

The MTC4BT was operated with W5500 module and Ethernet connection.

Mattzobricks, 2025-02-04:
Status: the problem still exists.

Mattzobricks, 2025-02-18:
I have a train on which the problems is 100% reproducible: the BR146 regional express.

The train has two PU hubs with one motor each, one in the loco in the front, and one in the steering car on the opposite side of the train. No lights. The configuration has not been changed significantly since a long time, and worked quite well.

After switching on both PU hubs, only one drives. The LED on the other one flashes 3 times, and even complies to the event triggers, that turn the LED white, red or blue (see configuration).

So far, the problem could be solved by switching off the hub in the steering car, and restart it. Then it always worked. This is no longer the case with when using the nimble 2.x branch.

Here is the loco configuration:

```
{
	"address": 146,
	"name": "BR146",
	"pwrIncStep": 10,
	"pwrDecStep": 10,
	"bleHubs": [
		{
			"type": "PU",
			"address": "90:84:2b:16:15:f8",
			"channels": [
				{
					"channel": "LED"
				},
				{
					"channel": "A",
					"attachedDevice": "motor",
					"direction": "reverse"
				}
			]
		},
		{
			"type": "PU",
			"address": "90:84:2b:00:5d:bb",
			"channels": [
				{
					"channel": "LED"
				},
				{
					"channel": "B",
					"attachedDevice": "motor",
					"direction": "forward"
				}
			]
		}
	],
	"events": [
		{
			"triggers": [
				{
					"source": "loco",
					"eventType": "dirchanged",
					"value": "forward"
				}
			],
			"actions": [
				{
					"address": "90:84:2b:16:15:f8",
					"channel": "LED",
					"color": "white"
				},
				{
					"address": "90:84:2b:00:5d:bb",
					"channel": "LED",
					"color": "red"
				}
			]
		},
		{
			"triggers": [
				{
					"source": "loco",
					"eventType": "dirchanged",
					"value": "backward"
				}
			],
			"actions": [
				{
					"address": "90:84:2b:16:15:f8",
					"channel": "LED",
					"color": "red"
				},
				{
					"address": "90:84:2b:00:5d:bb",
					"channel": "LED",
					"color": "white"
				}
			]
		},
		{
			"triggers": [
				{
					"source": "loco",
					"eventType": "dirchanged",
					"value": "stopped"
				}
			],
			"actions": [
				{
					"address": "90:84:2b:16:15:f8",
					"channel": "LED",
					"color": "blue"
				},
				{
					"address": "90:84:2b:00:5d:bb",
					"channel": "LED",
					"color": "blue"
				}
			]
		}
	]
}
```

Mattzobricks, 2025-02-19:
I experienced that the problem now also exists with my ICE. That's a commuter train with SBricks. So it is more and more doubtful that the problem arises from the LEGO PU hub stack.