Mattzobricks/MattzoControllers #68: Mouldking

DJChrisRey, 2024-02-16:
Hallo im forum wurde ja wegen mouldking geredet ich habe nun mehre akkus und motoren von mould king anbei link um das protokoll anzusprechen mit bis zu 4 angeschlossenen geräten .

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/MouldKg_nrf24l01.ino

DJChrisRey, 2024-02-17:
ich habe auch dies hier gefunden https://www.espruino.com/LEGO+Power+Functions+Clone
ALso es wird eine Bluetooth verbindung genutzt cool ist am akku eben das man mehre sachen gleich anschließen kann und per usb aufladbar ;)

Mattzobricks, 2024-05-20:
Wir haben uns das angesehen. Wird vermutlich nicht so einfach werden, da das Protokol nicht offenliegt.

Ich schließe das Thema vorübergehend.

Sobald es neue Informationen gibt, können wir uns damit gern noch einmal beschäftigen.

closed 2024-05-20 by Mattzobricks