Mattzobricks/MattzoControllers #77: Some bugs and problems from a user...

Mattzobricks, 2024-07-10:
I have a few related comments and questions:
– The manual (MTC4BT) talks about making a ‘platformio.ini’ file, but looking at the code in the existing platformio.ini file I assume this should be: make a ‘my_platformio.ini’ file. This given that the existing platformio.ini file expects a my_platformio.ini file. Or should I completely replace the existing platformio.ini file?
– The manual assumes the environment is by default set ‘az-delivery-devkit-v4’, but in v1.1 it seems the default is ‘esp32doit-devkit-v1’, so I guess it should be changed.
– I have a default error in the code of platformio.ini: ‘No option ‘additional_build_flags’ in section: [common]’. How do I fix this?
– In the platformIO tab on the left side of VS code, it always asks me to select a project folder. It repeats itself with this question after selecting the …\MTC4BT subfolder. How do I fix this?

Hilbert70, 2024-07-22:
I fixed:
– I have a default error in the code of platformio.ini: ‘No option ‘additional_build_flags’ in section: [common]’. How do I fix this?
In develop and added some comment around the my_platformio.ini and that you should rename the given example and why.

Hilbert70, 2024-10-03:
_– The manual (MTC4BT) talks about making a ‘platformio.ini’ file, but looking at the code in the existing platformio.ini file I assume this should be: make a ‘my_platformio.ini’ file. This given that the existing platformio.ini file expects a my_platformio.ini file. Or should I completely replace the existing platformio.ini file?_
Please change the documentation into copying the `my_platformio.ini.example` to `my_platformio.ini`.

_– In the platformIO tab on the left side of VS code, it always asks me to select a project folder. It repeats itself with this question after selecting the …\MTC4BT subfolder. How do I fix this?_
Use the file `MattzoControllers.code-workspace` in the root of the repository in VS code "File->Open workspace from file..."

_– I have a default error in the code of platformio.ini: ‘No option ‘additional_build_flags’ in section: [common]’. How do I fix this?_
This is fixed in the devolopment branch and changing the documentation to copy the my_platformio.ini.expample part.

Mattzobricks, 2024-10-10:
I changed the documentation of the MTC4BT like this:

"Some specifics of your ESP-32 and the connection between the PC to the ESP-32 are stored in the file platformio.ini. This file is under source code control, so changing it is not a good practice. For this reason, we are referencing a separate configuration file called my_platformio.ini from the platformio.ini file, This file can be changed freely at your choice. To get you started, we have created an example platformio.ini file for your convenience. The only thing you have to do is copy my_platformio.example.ini to my_platformio.ini. You then can open and modify it."

Mattzobricks, 2024-10-10:
Posted Hilberts answers in the forum. Thanks, Hilbert!

closed 2024-10-10 by Mattzobricks