Mattzobricks/MattzoControllers #78: error: 'e' was not declared in this scope

clawdelle, 2024-07-19:
Hi, sorry I'm to bad with this. Please, can somebody help.

mac m1, latest github source

Compiling .pio/build/az-delivery-devkit-v4/libcf2/MController/MCPinController.cpp.o
/Users/xx/mattzo/MattzoControllers/src/lib/MCNetwork/MattzoMQTTSubscriber.cpp: In static member function 'static void MattzoMQTTSubscriber::taskLoop(void*)':
/Users/xx/mattzo/MattzoControllers/src/lib/MCNetwork/MattzoMQTTSubscriber.cpp:139:19: error: exception handling disabled, use -fexceptions to enable
             throw message;
                   ^~~~~~~
/Users/xx/mattzo/MattzoControllers/src/lib/MCNetwork/MattzoMQTTSubscriber.cpp:153:63: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                 sendMessage("roc2bricks/ping", _subscriberName);
                                                               ^
/Users/xx/mattzo/MattzoControllers/src/lib/MCNetwork/MattzoMQTTSubscriber.cpp:163:58: error: 'e' was not declared in this scope
         log4MC::vlogf(LOG_EMERG, "Caught exception: %s", e.what());
                                                          ^
/Users/xx/mattzo/MattzoControllers/src/lib/MCNetwork/MattzoMQTTSubscriber.cpp:162:7: error: '...' handler must be the last handler for its try block [-fpermissive]
     } catch (const std::exception &e) {
       ^~~~~
*** [.pio/build/az-delivery-devkit-v4/lib0fb/MCNetwork/MattzoMQTTSubscriber.cpp.o] Error 1

Hilbert70, 2024-07-22:
How did you open the project?
Do you have the platformio extension enambled is vscode?
I would advise to use: Open Workspace from file...
Rename my_platformio.ini.example to my_platformio.ini
Select from the bottom your hardware and do a build.
I just tested the development branch with these actions and it compiled just file.

Hilbert70, 2024-10-03:
No feedback from the user, not reproducible, closing the  issue.

closed 2024-10-03 by Hilbert70