Mattzobricks/MattzoControllers #106: Emergency brake review

Mattzobricks, 2025-01-04:
I was wondering what the best action is to execute an emergency halt on Mattzobricks controlled layouts. There are numerous reasons for an emergency halt:
- Rocrail has detected a ghost train
- A critical controller (MTC4PF, MTC4BT, MLCs configurations with sensors, switches and bridges) disconnects from MQTT for whatever reason and a last will is broadcasted.
- MLC detects a signal overshoot and generates an ebrake event on MQTT.
- User clicks on "power down" or "emergency brake" on the Rocview UI.
- User clicks on a connected PU remote on the "ebrake" button.
- (add here if I forgot something).

When that happens, I have presently two problems:
1. The power button in Rocview is still "on". So I do not see immediately what happened and in what state the layout is.
2. I do not see the source of the problem.

How can we improve that?

joosbuijsNL, 2025-01-04:
As far as I know this is intended behavior of rocrail. 

I believe you can define an action in rocrail that powers down the layout on ebrake. 

Not sure if we can put an error like message on mqtt for rocrail to pick up and show in the console, this could/should indicate the reason for the ebrake if MTC4BT, MTC4PU or MLC triggers the ebrake.

Mattzobricks, 2025-01-04:
> I believe you can define an action in rocrail that powers down the layout on ebrake.

I thought so as well, but how? I looked, but I didn't find it.

Hilbert70, 2025-01-07:
Guys please look and try to understand https://wiki.rocrail.net/rocrail-snapshot/rocrail/wrapper-en.html

Look at sys, for giggles try `<sys cmd="shutdown" />` but to stop the laylout, just try `<sys cmd="stop" />` is this what you are looking for?