Mattzobricks/MattzoControllers #108: Stop train if PU remote disconnects

Mattzobricks, 2025-01-06:
If a PU remote looses its Bluetooth connection to the MTC4BT, the MTC4BT shall execute a "stop" command for all locos that it is controlling. If in list mode, that comprises not only the selected locomotive, but all locos in the list.

Hilbert70, 2025-01-29:
NOTE: only locos you have controlled/selected in the list in list mode are stopped. In free mode they are all stopped.

Mattzobricks, 2025-01-29:
I didn't quite understand the note:

"NOTE: only locos you have controlled/selected in the list in list mode are stopped. In free mode they are all stopped."

What do you mean exactly? Are you describing a deviation from the stated requirement?

I tested the following:
- One MTC4BT controls 2 pu remotes (pu1 and pu2).
- 3 trains running, 2 are controlled in list mode with pu1, 1 is controlled in liste mode with pu2.
- Disconnected the PU remote (schlepped it far away, so it disconnected from the MTC4BT).
- All 3 trains are still running. Nothing stopped.

Test negative.

Did I do something wrong?

Hilbert70, 2025-01-30:
The note means: when you are in list mode, you are able to select a locomotive. All locomotives that you have actively selected (possibly changed the speed) are stopen.
The ones you have never selected (pu remote did not change to the led colour of that item) those locomotive will not stop, reason you never actually controlled that locomotive so something else other than this pu is controlling it so on connection loss it may not stop that locomotive.

Mattzobricks, 2025-01-30:
Got it, makes sense.

Next step: test free mode.

Mattzobricks, 2025-01-31:
Tests in free mode executed. Everything fine.

closed 2025-01-31 by Mattzobricks