Mattzobricks/MattzoControllers #110: Improve speed steps for PU remote

Mattzobricks, 2025-01-07:
My 8wide E03 has a minimum "speed" of 50%. This means, that I need to press A+ 5 times that the loco moves at all.

Need to think about how we make the steps on the remote more usable. Maybe a mapping table step -> power could help.

Details to be discussed.

Hilbert70, 2025-01-10:
We get the lc command on the info topic with the (partly) the following information:
```
<lc id="FUX" addr="6337" prev_id="FUX" shortid="" 
...
V_min="30" V_mid="50" V_cru="75" V_max="100" V_maxsec="14"
KMH_min="0" KMH_mid="0" KMH_cru="0" KMH_max="0"
KMH_Rmin="0" KMH_Rmid="0" KMH_Rcru="0" KMH_Rmax="0"
KMH_Smin="0" KMH_Smid="0" KMH_Scru="0" KMH_Smax="0"
V_Rmin="0" V_Rmid="0" V_Rcru="0" V_Rmax="0"
V_Smin="0" V_Smid="0" V_Scru="0" V_Smax="0"
V_step="0" mass="0" minstep="0" maxstep="0" pwm="0" pwmcorrdiv="10"
Vmidpercent="30" Vmaxmin="20" Vmaxmax="255" Vmaxkmh="0" Vmidset="true"
V_mode="kmh" invdir="false" polarisation="true" regulated="true" restorefx="false"
```

There is a `V_min="30" V_mid="50" V_cru="75" V_max="100` and a `V_step="0"` (and a few more doing with speed) when filling in the right values in rocrail, the remote could use these.

This would be the preferred method because we do the locomotive configuration in one place

Mattzobricks, 2025-01-10:
Maybe we could make different speed-step configurations options for a loco available. You could either configure the remote with the traditional +/- 10% steps, or use I, II, III and IV (V_min, V_mid, V_cru, V_max) instead. What do you think?

Mattzobricks, 2025-01-29:
Duplicate to #127, closing issue.

closed 2025-01-29 by Mattzobricks