Mattzobricks/MattzoControllers #111: Loco does not respond if another loco is also configured in MTC4BT

Mattzobricks, 2025-01-07:
Tested with commit c7c3e83e82731e08393acd34215742ccad2b477a.

Configured two locos in the MTC4BT:
* ICE (2x SBrick)
* E03-8W (BuWizz2)

See attached configuration files.

Problem:
E03 does not respond if uploaded together with ICE loco configuration.

Synopsis:
After connecting the loco to the MTC4BT, the loco lights start to flash. This is shown in the serial log:

[0030] [0] [INFO] BLE : Scanning for 3 hub(s) aborted.
[0031] [0] [INFO] BLE : Connecting to hub '50:fa:ab:3d:86:90'...
[0032] [0] [INFO] BLE : Connected to hub '50:fa:ab:3d:86:90'.
[0033] [0] [INFO] BLE : BuWizz2 : Watchdog timeout not set
[0034] [0] [INFO] Loop: Connected to all hubs of loco 'E03-8W'.
[0035] [0] [DEBUG] Set power level 03
[0036] [0] [INFO] BLE : Scanning for 2 hub(s)...

After BLE connection, the loco can not be brought into operational mode. The lights continue to flash. Sending a "power on" event does also not help.

The configuration of the loco works fine if uploaded alone (without the ICE loco configuration).

The ICE train works alone and also together with E03 as in the attached configuration.

Tested also with commit 5608dd90921eb79deb829c2d02dd5dda28b4d9e3. With this commit, everything works fine. Both locos respond and can be controlled normally.

Tested multiple times with stable results.

Mattzobricks, 2025-01-07:
Further testing reveals:
Commit 76e8659fffe3bbfdbf180357f26c612e37168831 works.
Commit 1fedba39b20479ee431c4a7ae7033bb9bbec6ab6 does not work, so it likely introduced the bug.

Mattzobricks, 2025-01-07:
To be more precise, it is precisely this change that generates the bug:

Before (works):
```
   if (freeConfig["button"].is<std::string>() &&
            freeConfig["type"].is<std::string>() &&
            freeConfig["action"].is<std::string>() &&
            (freeConfig["id"].is<const char *>() || freeConfig["addr"].is<int>())) {
```
            
After (does not work):
```
            if (freeConfig["button"].is<std::string>() &&
                freeConfig["type"].is<std::string>() &&
                freeConfig["action"].is<std::string>()) {
```

Hilbert70, 2025-01-07:
It has to do with the removal of the warnings and making a {} of the two case statements.

At the moment I have a remote and a pu hub with the same led colour without setting one, only occurs after a reset of the ESP, will continue the search.

Hilbert70, 2025-01-07:
Should be solved in develop, something with uninitiated variables :'(

Mattzobricks, 2025-01-09:
The bug disappeared.

closed 2025-01-09 by Mattzobricks