Mattzobricks/MattzoControllers #119: Incorrect speed handling after power is back on

Mattzobricks, 2025-01-26:
The target speed model of the MTC4BT is not implemented correctly.

When the power goes off, the train stops immediately. This is correct.

When the power comes back on, the train leaps immediately back to the old speed instead of accelerating gently.

Problems:
- This doesn't look good.
- For long and heavy trains with a strong loco that means, that the magnets of the loco will disconnect from the waggons.
- Brutal accelerations can caused derailments, especially with heavy trains in tight curves.

Additional problem: when the target speed was set to another value (let's say "0") while the power was off, and the power comes back on, the train will brutally accelerate to the old speed, and then slowly come to a stop with the defined power decrement.

What it required is rather, that the target speed for the loco is constantly updated in the controller, and after the power comes back, there is a gentle acceleration to the target speed. If the new target speed is 0 in between, the trains must not move at all when power comes back.

Hilbert70, 2025-01-29:
Please retest the develop branch, I could not reproduce this (with mentioned commits)
```
[0279] [0] [DEBUG] Current power: 94
[0280] [0] [DEBUG] Current power: -98
[0281] [0] [DEBUG] Current power: 98
[0282] [1] [INFO] MQTT: Received 'stop' command. Stopping all locos.
[0283] [1] [DEBUG] Ctrl: Received lc command for loco address '2', speed 0.
[0284] [1] [DEBUG] Ctrl: Received lc command for loco address '2', speed 0.
[0285] [1] [INFO] MQTT: Received 'go' command. Releasing e-brake and resuming all locos.
```
After this no `Current power` debug lines and the motors have zero speed.

Mattzobricks, 2025-01-29:
Successfully tested. Thanks, @Hilbert70.

closed 2025-01-29 by Mattzobricks