Mattzobricks/MattzoControllers #120: Can not connect more than 5 BLE hubs to MTC4BT

Mattzobricks, 2025-01-26:
At present, you can not connect more than 5 BLE hubs to the MTC4BT. Tested with commit b41b15bef05f7cfc5186cd735ea2c22e5c490d8b.

When disconnecting the fifth hub, and connect and 6th hub, no. 6 does not respond to rocrail commands (i.e., the loco won’t go). When disconnecting it, and reconnecting hub no. 5 again, the controller crashes.

Video: https://www.youtube.com/watch?v=OQSTGpovb0g

Hilbert70, 2025-01-28:
Fixed in develop, only for the seeed_xiao_esp32-s3 board

Mattzobricks, 2025-01-29:
Thanks for introducing a present maximum number of 5 hubs for the Seeed Xiao ESP32-S3 board.

Next step would be to solve the problem that only 5 hubs can connect.