Mattzobricks/MattzoControllers #124: Bad structure network_config.json

Mattzobricks, 2025-01-28:
This is an example for a present network_config.json structure:

```
{
	"logging": {
		"min_level": "info",
		"serial": {
			"enabled": true
		},
		"syslog": {
			"enabled": false,
			"server": "mattzonas",
			"port": 514,
			"appname": "MTC4BT"
		}
	},
	"network" : {
		"hostname": "mtc4bt-stat-e1",
		"otaPassword": "mattzobricks",
		"type" : "waveshare-ESP32-S3-ETH"
	},
	"mqtt": {
		"broker": "192.168.178.23",
		"port": 1883,
		"keepalive": 10,
		"ping": 0
	}
}
```

Deficiancies: 
1. hostname should be transferred to controller_config.json, as it varies from controller to controller.
2. The "type" should be reserved for the physical network type (```wired```, ```wireless```). Cross-dimensional, even hardware-specific entries like "waveshare-ESP32-S3-ETH" disrupt the structure, and lead to all well-known follow-up problems in software development. The specific "board" that is used in the configuration shall be transferred in controller_config.json, or rather be automatically set by the used platform.io environment (if that's possible).

Hilbert70, 2025-01-28:
1. "hostname" should be in a separate file, or stay in network or in a new file.
 2. "type" is now "wired" or "wireless" you **have to** select the right board in `platformio.ini` to select the correct wiring of the W5500 module.

Mattzobricks, 2025-01-28:
ad 2: great!

ad 1: I will setup a call and explain again, why hostname should be in controller_config.json.

closed 2025-01-28 by Hilbert70

Mattzobricks, 2025-01-29:
From the call of yesterday evening:
- hardware specific configuration is now kept outside network_config.json (thanks, Hilbert).
- the hostname is still in network_config.json, which is not desired for the stated reasons. We will tackle this issue again when we pimp the configuration of the controller for web- or net configuration. Maybe the problem resolves itself after that.