Mattzobricks/MattzoControllers #126: MTC4BT considers all power settings below V_MIN as 0

Mattzobricks, 2025-01-28:
The MTC4BT considers all power settings below V_MIN as 0.

Example: V_MIN is 30%, the user chooses a speed of 10%. The loco does nothing.

This is annoying especially when trying to move a train with the PU remote. Some trains have v_mins in the area of 80%. That means that you need to click 8 times on the + button to make the train move.

If the remote should rather set v_min, v_mid, v_cruise and v_max in the future instead of 10% increments, is another question and out of scope of this issue.

Hilbert70, 2025-01-29:
If v_min is set honor it, the same goes for (if non zero):
V_min="30" 
V_Rmin="0"
V_Smin="0"
```

R (Reverse) and S (shunting)?

Mattzobricks, 2025-01-29:
I don't agree. v_min, V_Rmin and V_Smin should not play a role.

Look at this example (and there are many more use cases): You need to set v_min to 50%, because else the locos with 10 cars behind won't move through curves. If the loco is decoupled, 50% is way to high.

The v_min makes no sense if the user demands it otherwise. This is implemented correctly in MTC4PF, and we should also do it in MTC4BT.

Mattzobricks, 2025-01-29:
The situation is even worse than described. When you go with v_min (let's say: 30%), and reduce the speed by 10%, the loco still goes with 30% power. Same when reduce one step more (10%), it still goes with 30%. Only after you reach 0%, it stops.

Mattzobricks, 2025-01-29:
As the situation is really grave in my mind and destroys a lot of fun when playing with the remote, I had a look at the code myself and fixed the problem. Operations under v_min are now possible.

I also found out, that trains didn't stop immediately when the target speed was set to 0. They decelerated down under v_min first, and only then stopped. This explains a good dozen of severe crashes that I had during the last years. Fixed that as well.

I'll try to be a good boy and create a PR.

closed 2025-01-31 by Hilbert70

Mattzobricks, 2025-01-31:
Regression test for MTC4BT done. Works great.