Mattzobricks/MattzoControllers #131: Combination of loco and remote does not work

Mattzobricks, 2025-01-31:
I setup one classic ESP32 (wired) for a Buwizz2 unit in my E03 / 8-wide and a PU remote control. I wanted to do a short video about the BuWizz support in the firmware release, but it didn't work in this combination. Find the configuration attached. When I connected the Buwizz2 unit to the controller, the train lights continue to flash, and I couldn't trigger any motor movement, not with the remote, nor with Rocrail directly. The connected remote works and does its job, though. You can see the power changes and function button presses in Rocrail.

When I remove the remote, and configure the loco alone, it works.

Hilbert70, 2025-01-31:
Added a delay of 250ms after an mqtt message is send when a remote button is pressed. And made adding events in the code more robust.

Mattzobricks, 2025-02-02:
Reminder: this change is in the develop branch. Before releasing 1.2, Mattze would like to understand the change better.

Hilbert70, 2025-02-03:
After changes (before adding a delay), could not reproduce.

Mattzobricks, 2025-02-04:
Added the problematic config as an example configuration (decef3fc1aac95e48dab452180244506c800328b).

Mattzobricks, 2025-02-10:
No longer reproducible after Buwizz2 firmware upgrade from 1.0.30 to 1.2.30.

closed 2025-02-10 by Mattzobricks