Mattzobricks/MattzoControllers #132: Allow PWM values between 500 and 2500 mUs for servo control in MLC

Mattzobricks, 2025-02-02:
When updating my form distant signals (which require relatively large servo angles), I found out that the PWM value range for servo control in the servo library of platform.io spreads from 1000 to 2000 mUs (microseconds) only. This must have changed during the last 2 or 3 years.

In Arduino, the range is 500 to 2500 mUs, which is better.

As we are using 270° Geekservos, which have a range from 600 to 2400, we need to extend the range like it is in Arduino.

Hilbert70, 2025-02-02:
A better way is to use the attach(int pin, uint16_t minUs, uint16_t maxUs, int value) function,

```
 _maxUs = max((uint16_t)250, min((uint16_t)3000, maxUs));
  _minUs = max((uint16_t)200, min(_maxUs, minUs));
```

This overrides the constructor values:
```
 _minUs = DEFAULT_MIN_PULSE_WIDTH;
  _maxUs = DEFAULT_MAX_PULSE_WIDTH;
```

Mattzobricks, 2025-02-02:
Sure, I did that.

closed 2025-02-02 by Mattzobricks