Mattzobricks/MattzoControllers #140: Improved power scale (min/max)

Mattzobricks, 2025-02-12:
As in the MTC4PF, there is a useful range of power settings. We should introduce that in the MTC4BT as well.

I will implement it myself. This will be less effort, because it needs to be tested with a large variety of different Bluetooth trains, and the example config files also need updates afterwards.

Mattzobricks, 2025-02-14:
I had a look at the code in MTC4BT. It appears that there is no abstraction of loco speed and power setting as in the MTC4PF. This abstraction would be required to translate a gently speed change to an appropriate power range in the controller, or at least it would faciliate things significantly. Can't be done swiftly, and needs some systematic testing, so I removed the feature form V1.2.1.

Hilbert70, 2025-02-18:
All speed related calculations are done in:
```
bool MCChannelController::UpdateCurrentPwrPerc()
```
in `src/MTC4BT/lib/MController/MCChannelController.cpp`

Please do not abstract more, it is already quite deep in the code.

Mattzobricks, 2025-02-18:
Have a look at the MTC4PF code. The code for speed and power handling is a fraction of the MTC4BT code, and it works.

If you do things right, things can be done easily and effectively.