Mattzobricks/MattzoControllers #146: PU remote does not work if no loco is configured in Rocrail

Mattze0815, 2025-02-25:
When there is no loco in RocRail configured, the BT remote does not work.
Tested in free mode
Not tested in list mode

Requirement is that the PU remote works in both modes, even if there are no locos. The ratio is that one could just configure a layout and use the remotes to control switches and signals, but the trains are not controlled by Rocrail.

Hilbert70, 2025-02-25:
**NOTE** if you have loco's in your list, the LED could stay white and you will get the debug messages:
```
[0352] [1] [DEBUG] Going for delete loco list.
[0353] [1] [DEBUG] Going for loco list.
[0354] [1] [DEBUG] Got for loco list.
```

This is expected and wanted behavior!

ogel0815, 2025-02-25:
Works fine

commit df9eef29 (2025-03-03): Fixed #146
closed 2025-02-25 by ogel0815