Mattzobricks/MattzoControllers #148: Proposal: get change firmware/configuration via a simple http server triuggered by an mqtt message

Hilbert70, 2025-02-27:
# Use case update configuration

Using the web download method, it adds files to the ESP filesystem (SPIFFS), or over writes existing ones!

## Update to a new controller configuration

During development, or running locomotives on a fair, you don't want to startup vscode, or issue an `pio` command to update your configuration, or switch to a new configuration all together.

Since you have your development laptop handy, running probably `mosquitto` and `rocrail`, you have the ability to send an mqtt message. Wouldn't it be nice to send, lets say `update controller config new_fancy_config` to the topic `mattzobricks/mtc4bt_001`, where `mtc5bt_001` is the controller hostname. This command will only work if we introduce a new configuration item `baseURL`, and have a simpel http server running:
```
python3 -m http.server 8888
```
would be enough to have a simple web server running local.

When this command is received, the mtc4bt connroller with hostname `mtc4bt_001` will look in then `<baseURL>/configs/new_fancy_config` will donwload `controller_config.json`, scan this file for additional configuration files and download these, do a reboot and the new configuration will be active.

## Update network settings, aka update `network_controller.json`

This is quite similar, but a bit more dangerous.
Let say we issue the `update network config` on the topic `mattzobricks/mtc4bt_001`, this will command the ESP into looking for the file `network_config.json` in the map `<baseURL>/mtc4bt_001/network_controller.json`, if is is there it will download it and do a reboot, using this new configuration. Since the hostname is also in this file, you could set a new hostname for this controller, hence the dangerous part.

To remedy  this danger, before downloading this file, it will create the `hostname` file on the ESP and put the original hostname into that file, and during startup, if this file exists, it will overrule all hostname references in the other configuration files.

This construction will also open up the possibility to have a common `network_config.json`, to let the ESP use that one, we could issue the `update network common config`, this will instruct the ESP to look into the map `<baseURL>/common/network_controller.json` download it and reboot.

## roll back to your 'OTA' config

If you want to revert back to the old config stored in the directory `data_dir` just start vscode and do the `Upload Filesystem Image` this will clear all files updated with the mqtt commands.

# update firmware **DANGEROUS**

**ONLY FOR POWER USERS**

During development, you may want to switch between firmware versions, to test things, or revert back/forward to a stable one. And don't want to do all kind of git stuff and compile firmware version.

We can again issue an mqtt command, lets say `switch firmware Some_test_version` to the topic `mattzobricks/mtc4bt_001`, the ESP knows processor it has, lets say esp32, then it will try and download from `<baseURL>/esp32.firmware/Some_test_version.bin` after a success full download, it will reboot and have `Some_test_version` as its firmware.

To make this usefull, we need to somehow display the firmware name, or request it via an mqtt message. Say `show firmware version` on the  `mattzobricks/mtc4bt_001` topic. This should post the `mtc4bt_001 has version "Some_test_version" message to the `mattzobricks/info` topic.

If the `show firmware version` is posted at the `mattzobricks/command` topic, all MTC4BT's on the network should report back their version.

**Prerequisite** all compatible MTC4BT firmwares hust have knowledge about their version.

A way to make this happen is to introduce some magic into the building of the firmwares, something like this, in the `platformio.ini` the following must be introduced:
```
extra_scripts = 
    pre:../bin/auto_firmware_version.py
```
This file contains the following:
```
import subprocess
import os

Import("env")

def get_firmware_specifier_build_flag():
    ret = subprocess.run(["git", "describe"], stdout=subprocess.PIPE, text=True) #Uses only annotated tags
    build_version = ret.stdout.strip()
    build_flag = "-D AUTO_VERSION=\\\"" + build_version + "\\\""
    print ("Firmware Revision: " + build_version)
    return (build_flag)

env.Append(
    BUILD_FLAGS=[get_firmware_specifier_build_flag()]
)
```

and in the source code 
```
log4MC::vlogf(LOG_INFO, "\nMTC4BTversion %s", AUTO_VERSION);
```
and the (global) variable
```
char * firmware_version = AUTO_VERSION;
```
should be set.

And in the development branch we should use for example set the tag to `D1.3` or whatever version we are working on. In the development branch we should set `V1.2.1` the current version.

Then the version will be something like: `V1.2-RC2-42-g7c325cb` (this is the output from  `git describe` at the 27th of February 2025).

An other mechanism must be used when `git describe` is empty, in case of zipped releases, then the python script should use the directory name for our current release the version should report back `MattzoConrtollers-1.2.1`

The side benefit is that we can now ask users for the version so we know which firmware they are using.

## roll back

Just do the firmware compilation an upload it to your ESP

Hilbert70, 2025-02-27:
## version "numbers"

An other version name for development versions could be:
`git branch --show-current`  -> `develop`
and add to that 
`git rev-parse --short HEAD` -> `7c325cb`
For master version stick with with the tagged version, which should be right!

ogel0815, 2025-02-27:
I'm not very keen on dismantling my controllers just to update them by wire. With OTA only one controller can be updated at a time and I need vs code. If I have to change the network configuration globally because I'm going to an exhibition or something has changed in my home, it would be great to be able to tell the ESPs that they should all load the new network configuration themselves (broadcast?). It would be great to be able to say that there is a new, well-tested firmware and that it should be downloaded by everyone who is affected with respect to its hardware.
It should be possible to prevent unnecessary updates by using checksums or known version numbers. Then a command like “mattzobricks/check4updates” could cover everything. Well, a script will be nessesary to send a message to each of all known controllers.

As you have already written, the configuration files for each controller can be stored in a structured way (with directories or naming conventions).

The risk of overwriting the host name mentioned can possibly be prevented if it is stored in a separate configuration file. I have not yet looked into this.

Yes, there is a certain risk involved.
Yes, this is a great idea.